bigtreetech/BIGTREETECH-TFT24-V1.1 #1: FW source files needed

ArturNadolski, 2019-06-28:
Please upload FW source files.

Msq001, 2019-07-10:
will do this，  I'm working on integrating 3.5, 2.8 and 2.4 inches in one source code, and just need modifying config to generate correct firmware, It is scheduled to be completed before 2019.7.16 (china time), and I will upload the source code then.

ArturNadolski, 2019-07-10:
It's good idea to unify software platform for all displays.

DavidThijs, 2019-07-11:
Indeed, the source would be welcome so that we can tweak the display ! Thanks a bunch for the nice products that are developed !

Msq001, 2019-07-16:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware  source has be released in here， it contains TFT35 V1.0 V1.1 V1.2 TFT28 TFT24 V1.1，you need only modify 
`env_default = BIGTREE_TFT24_V1_1` in platformio.ini

Talha909, 2019-12-04:
I have changed the BIGTREE_TFT24_V1_1 with old one now the compile give me the error.

closed 2019-12-18 by ArturNadolski