bigtreetech/BIGTREETECH-TFT24-V1.1 #6: How do I configure Marlin LCD12864 emulation

tiaanv, 2019-09-03:
I have a SKR PRO 1.1, and the TFT24-V1.1 board.  I cannot figure out what settings I need in Marlin for this to work.

 have updated to the latest firmware on the TFT24, but it just shows "LCD12864 Simulator"

What LCD should I enable in Marlin?
Do I need any other settings?

tiaanv, 2019-09-06:
Anybody There?

defalkner, 2019-09-14:
Push and hold the round button and other options will show up allowing you to switch to the touch screen interface.

tiaanv, 2019-09-14:
Of course I know that part. What I am asking is as what type of LCD so I need to config this in Marlin, as setting it to Full Graphic does not work.

kkaleta, 2019-09-17:
I've just figure it out
Ypu need to **uncomment** line
`#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`

https://pastebin.com/2GrL8gkF

myudshlihe, 2019-09-27:
What should I do with SKR-E3-DIP? 
If I choose REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER I get error message "Only CR10_STOCKDISPLAY supported for this board".
If I choose CR10_STOCKDISPLAY - firmware build ok. But after flashing I see splashscreen once and then it hung. After reset it just shows "LCD12864 Simulator".

myudshlihe, 2019-09-27:
I find it!
ENCODER_PULSES_PER_STEP by default was 3 in config example. It need to be changed to 4 and it work.

printin3dprints, 2020-04-23:
im trying to set up th skr mini e3 v1.2 on a printer i built with the same kind of set up as the anet a8, am i able o choose the anet a8 configuartion and then change the board the the skr mini e3 v1.2? and also i have the tft 24 v1.1, how do i configure it in marlin?

segunakino, 2020-09-16:
> Push and hold the round button and other options will show up allowing you to switch to the touch screen interface.

Really appreciate this simple instruction! It worked! Thanks!