bigtreetech/BIGTREETECH-TFT24-V1.1 #11: Config Preheat Settings

MrOmbre, 2019-11-24:
I just update to v1.1.24.2 with new fonts and bmps. I realized, that the "Preheat" Button is new and I am glad about. But where can I change the Preheat parameter for PLA, PETG and ABS ?
The Preheat parameter in LCD12864 works proper, but the settings will not take over to Touch Mode

MrOmbre, 2019-11-30:
anybody an idea?

Hagbard89, 2020-01-14:
you have to change it in the Marlin firmware

bobofthebush, 2020-01-25:
how do you mean change it in the marlin firmware. do you mean exit out to the marlin menu system to preheat and then switch back over to touchscreen version to print? Isn't there a way to set it up in the touch screen menu?

Hagbard89, 2020-01-25:
i didnt find a way to change it directly at the touchscreen, method 1: change it with a Fullgraphics display or a 2004 lcd and save storage. method 2: you can send a gcode to you printer to change these values. i use octoprint to do it, there is a terminal tap. https://marlinfw.org/docs/gcode/M145.html these link will help. pla is normaly material 1. Send "M145 S0 H200 B60 F0" S0 means materialstorage 1, H200 means Heater 200°C, B60 means bedtemp 60°C and F0 means fan at 0%, then send an m500 to store it in the eeprom of your printer. i hope that helps

MrOmbre, 2020-01-26:
If you send gcode to your printer with terminal like "M145 S0 H200 B60 F0" helps not in touch mode. You only can storage in marlin mode, not in touch mode. You can change preheat values in marlin mode in the menu configuartion > preheat, you don't have to send gcode to printer for. The preheat settings in touch mode can only be change befor compile to firmware.bin

I have no idea, why change preheat in touchmode is so complicated. I you want to change the value, you have to change in source code and compile to an .bin file, after flash again the firmware.
Only for preaheat changing, unbelievable

Hagbard89, 2020-01-26:
i' tested it, its true... you have to compile new tft firmware. sorry about that

Benderrodrigu3z, 2021-01-28:
The ANSWER is in the Config.ini file for the TFT screen