bigtreetech/BIGTREETECH-TFT24-V1.1 #12: Text in TFT mode glitched

Enoblk, 2019-12-03:
The text in TFT mode is not displaying correctly, I've flashed the new firmware bin that was preconfigured  as well as one I compiled 

[Images](https://imgur.com/a/SOZ7hBB)

sightlyowl, 2019-12-24:
You are missing files when you updated.
1.format a SD card
2.download this [BIGTREE_TFT24_V1.1.25.bin](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/archive/master.zip)

3.Extract the files

4.Find "BIGTREE_TFT24_V1.1.25.bin" and place that in the root directory of your SD card.

Copy to SD Card root directory to update
5.Find "tft24" folder that contains these two folders font & bmp. 
6.make sure your folder structure looks like this.
[image]
[image]
[image]
[image]

7.Once that is complete remove the SD card from your PC and install it in your TFT24 SD slot. 

8.Hit reset and every thing should start to update with the required images and fonts.

Migamix, 2020-01-11:
pleaase close this is if was solved for you. 
this is the correct answer to the best of my knowledge and was my solution also
include the font and bitmap folders and you should be solved.