bigtreetech/BIGTREETECH-TFT24-V1.1 #14: "Illegal FLASH APP" after firmware update

Pilliwiddlipillipitit, 2019-12-04:
After updating to a newer firmware (BIGTREE_TFT24_V1.1.25.bin) the warning "Illegal FLASH APP" is shown. I've read some other issues about this topic, but I can't find a clear solution for this issue.

Someone tried different SD-Card sizes and it worked. Also there is a solution by flashing a new bootloader. Is there a bootloader for this display or is it the same bootloader like in the other issue? But I don't understand how I destroyed the bootloader by updating the firmware. The bootloader should not destroy itself during this process.

However, is there a solution for my problem?
By the way, I have an ST-Link and it would be possible for me to reflash the bootloader.

Pilliwiddlipillipitit, 2019-12-05:
After several tests with different SD-Cards (2x 128MB and 1x 8GB) it was really the file format.
At my first attempts the SD-Cards were quick formatted to FAT32. This didn't work.
I don't know why, but somehow it worked after the SD-Card was fully formatted to FAT32 with default settings.

Hope this will help others.

Pilli

closed 2019-12-05 by Pilliwiddlipillipitit

jimprince, 2020-06-13:
Thanks, this solved my issue.