bigtreetech/BIGTREETECH-TFT24-V1.1 #15: Octoprint with TFT not working

manukieli, 2020-01-25:
I try to send printing from octoprint with the TFT mode and is not working.
Do nothing.
But with the LCD mode they work well.

Hagbard89, 2020-01-25:
what exactly is not working? does the tft mode work without octoprint connected?

manukieli, 2020-01-26:
Yes the TFT mode work well. But is I send a print from Octoprint, the printer do nothing.

blkhawk, 2020-06-19:
I guess I will answer since I know what the problem is. I fell for this myself. The BTT Firmware works different from what you would expect after using say octoprint. 
The Firmware will send all the settings on the display every second to the controller. This interrupts anything octoprint does. 
The firmware really needs a "Passive" mode.

BradNagel, 2020-06-26:
I just hit this after installing the TFT24 on my printer yesterday. Octoprint worked for 15 mins then stopped updating messages on progress. Yet still would receive temp for the nozzle and hot bed.