bigtreetech/BIGTREETECH-TFT24-V1.1 #22: All low Error message but screen works fine

w1n5t0ngr, 2020-06-13:
I've recently installed the tft24 to my skr 1.4 turbo with TMC2209 and configured the firmware according to my needs but there is a little bug that annoys me more than anything else. 
Every time I turn on the printer, it shows the message "printer is not connect" and after a couple of seconds it throws an error message (with a double buzzer noise) that reads "all low". After this strange behaviour, everything works as expected and the printer is detected correctly without any user interaction. I have tried removing the 2 exp cables but the problem persists. Any ideas?
PS: 
First serial port in marlin is configured to 0 and the 2nd one to -1.

[image]

mrgaryth, 2020-06-17:
I have this same problem. I have mine connected to SKR v1.3.

DavidThijs, 2020-06-18:
Mine does work fine in both 12864 emu and native mode on an SKR PRO 1.1. For the SKR, I needed to put the serial port 0 to -1 and the 2nd to 2. I suppose there are some similarities between these boards, it doesn't harm to give it a try.