bigtreetech/BIGTREETECH-TFT24-V1.1 #30: Mks Gen L v1.0

tonimonton, 2021-01-06:
can someone tell me if they have managed to get it to work with this motherboard?

DavidThijs, 2021-01-07:
I have to check my old configs/source but a tft24 surely works with the MKS GEN L1.0 even in combination with a 128x64 LCD. Allow me to check tomorrow evening.
Nowadays, I run a SKR PRO1.0 with a TFT35 and the SKR is a different beast.

kazuimis, 2021-07-10:
I am facing an issue as to uploading the Firmware when I put in the SD card. I could only update the and use the touchsceen mode at the moment. The LCD screen mode also does not see to work and I am not sure did I placed the ribbon cable wrongly? I have tried to twist the nobs too to ensure that it was not responding or smt. I hope I can fix this so I can too enjoy the dual mode! :D

I am using TFT24 to MKS Gen L on ArtillerySidewinder X1.
[image]
[image]
[image]

kazuimis, 2021-07-10:
Ah alright I kind if know how to get the LCD screen running from a youtube that happen to share on it. After that I went to download the latest TFT24 firmware and tried updating it and now I am stuck with this. :/