bigtreetech/BIGTREETECH-TFT24-V1.1 #31: TFT24 v1.1 | SKR 1.3 | TMC2130 | No printer attached

Wyso77, 2021-02-20:
Hi. Please help me configure the screen TFT 24 v1.1. For a week I have trying to configure it unsuccessfully. The problem is "No printer attached". I used various videos on YT, help on GitHub and nothing...

I have LCD: TFT 2.4 v1.1, board: SKR 1.3 with TMC1130 and I check (old and new) original LCD firmware and Marlin firmware (old and new - 020007). My settings:

#define SERIAL_PORT -1 (or mix like e.g 0)
#define SERIAL_PORT_2 0 (or mix like e.g uncomment or -1)
#define BAUDRATE 115200 (or 250000)

[image]
[image]

In screen is TMC2208 and reset pin is disconnected
I have no ideas how to fix it. Please help me.
Best regards.

radek8, 2021-04-11:
Did you solve your problem?

Wyso77, 2021-04-13:
> Did you solve your problem?

Hi, yes! You have already helped in this topic:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1636

I don't understand this question. 
All the best :).

closed 2021-04-13 by Wyso77