bigtreetech/BIGTREETECH-TFT24-V1.1 #35: "Nenhuma impressora conectada!" Erro com btt SKR 1.4 Turbo

Laidner, 2022-05-09:
Olá! Comprei o SKR1.4 Turbo + BTT-TFT24-V1.1 um tempo atrás e usei ambos para atualizar minha impressora Longer Lk4 Pro. Confesso que sou leigo e não entendo muito sobre essas configurações! Segui um vídeo: https://www.youtube.com/watch?v=gVMiEtfK9S8 para me auxiliar mas estou com um problema ao tentar me conectar que não consigo corrigir. Sempre aparece na tela "Nenhuma impressora conectada"

Minha configuração:
BTT SKR V1.4 Turbo
Marlim 2.0
BTT TFT24-V1.1

[image]
[image]
[image]

Por favor me ajude, obrigado!

Laidner, 2022-06-21:
Thank you for replying!
I used serial port 0 and -1
On Baudrate I tried 57600, 115200, 250000 on both Marlim and screen
I'm Brazilian, I followed this video
[image]
I'm Brazilian, I followed this video:
https://www.youtube.com/watch?v=gVMiEtfK9S8&t=1654s

I'll check and try to correct the error with this link you sent me!

radek8, 2022-06-21:
Does Merlin mode work for you?
Users with the same problem have written here several times.
There may be a port configuration or speed error. (you wrote that you checked)
The serial cable (black 5-pin) may also be defective. Check the cable !!

Another option is faulty serial port pins on the SKR board or on the TFT. (common defect see link)
If the pins on the SKR are defective, the pins from the wifi connector can be used. I need to activate the correct serial port in merlin.

If the pins on the TFT were defective, the pins from the EXP connector could be used. Then you would no longer be able to use the merlin mode, but it won't matter if the touch mode is functional :-)