bigtreetech/BIGTREETECH-TFT24-V1.1 #37: SKR 1.4 Turbo + TFT24 : Marlin Mode blank screen

mikebzh44, 2023-01-27:
Hello.

When I switch the TFT24 screen to Marlin mode, all I have is a blank screen with only Marlin Mode written at the top :(

I have checked my 2 IDC10 cables and it's straight cables

I have configured my marlin files following this : https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#marlin-dependencies
Except for the Z_STEPPER_AUTO_ALIGN as my 2 stepper motors are driven by the same driver.

My config files are attached here but I have to change the extension for the upload (.h to .txt)

Can you please help me ?

[Configuration.txt](https://github.com/bigtreetech/BIGTREETECH-TFT24-V1.1/files/10517471/Configuration.txt)
[Configuration_adv.txt](https://github.com/bigtreetech/BIGTREETECH-TFT24-V1.1/files/10517473/Configuration_adv.txt)

radek8, 2023-06-15:
The display configuration in Marlin looks fine.
It's probably a HW problem.

radek8, 2023-06-15:
Are the EXP1 and EXP2 connectors oriented correctly?

mikebzh44, 2023-06-15:
Hello.

Thanks for your reply but I have switch to Klipper so I don't use this screen anymore.

closed 2023-06-15 by mikebzh44