bigtreetech/BIGTREETECH-TFT24-V1.1 #38: TFT 2.4 v1.1 Marlin keep restarting

El-kady, 2023-02-19:
Hello,

I'm using TFT 2.4 v1.1 with arduino mega and ramps 1.4 (1.6 too)
it works fine on USB power only
when i connect it to power supply it keeps restarting for ever
i disconnected the TFT and connected normal ramps screen and it working

what is that mean? is that software issue or a hardware issue

tomtom25000, 2023-02-22:
Hello,

I have the same problem with the same equipment,
I haven't updated the tft yet, I just bought it.

Did you update the tft?

tomtom25000, 2023-02-23:
I updated and still the same.
I fixed the problem by adding an ams1117 5V regulator to power the TFT AND lowering the brightness of the TFT.
I wonder if it is necessary to put a logic level shifter between the TFT(RS232) and the card ramps .

El-kady, 2023-02-23:
I solved the problem by removing the 5v regulator from the arduino and add another one at the back between 5v - ground - vin

check the images

[image]
[image]