bigtreetech/BIGTREETECH-TFT24-V1.1 #41: TFT 24 v1.1 lost RX

Akeyr, 2023-06-07:
Hi, have a very specific issue, my TFT 24 has lost RX absolutely. The display can not recieve signals from the motherboard (skr mini e3 v2).  I can move axis through a display, but nothing else. Reflashed TFT and SKR many times, useing the last firmware, but nothing helped)

radek8, 2023-06-15:
A TFT processor has multiple serial pins.
If you don't mind losing the simulated Marlin mode, the pins in the EXP Connectors could be used. After a little FW modification in TFT, touch mode would work.
Someone here has already done this successfully in the past.

It would be possible to get the Marli mode working by rewiring, but that would be more complicated and I don't know how skilled you are... :-)

radek8, 2023-06-15:
One more note.
There are two possible causes of your problem
Faulty RX pin on TFT, or faulty TX pin on SKR.
Are you sure the problem is on the TFT?
Can you test the display on another printer or connect another display to your printer?

radek8, 2023-06-15:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2382#issuecomment-1035084997

Akeyr, 2023-06-16:
> [bigtreetech/BIGTREETECH-TouchScreenFirmware#2382 (comment)](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2382#issuecomment-1035084997)

That looks helpfull, very thankyou)

Akeyr, 2023-06-16:
> A TFT processor has multiple serial pins. If you don't mind losing the simulated Marlin mode, the pins in the EXP Connectors could be used. After a little FW modification in TFT, touch mode would [...]

I am very skilled in 3d-printer manufacturing, and absolutely dumb in programing)))) But i think i will be able to make some small changes in TFT firmware to make it work as you adviced. And i tryied to connect this display to another motherboards, and yes, the problem is in TFT. It works the same way with other motherboards (skr 1.3, mks sgen-l, skr 1.4, skr 2 - all the same)

radek8, 2023-06-16:
Your TFT has a faulty pin on the TFT connector.
So remap the serial connector and connect it to the EXP connector.
If the display will work, you can try to remap the EXP connector (According to the instructions in the link).
If one pin from the TFT connector was functional and the pin from the beeper was not connected, Marlin mode could also work after further FW modification. :-)
In the evening I will see which pins are affected and if it would be possible to remap and replace them...

radek8, 2023-06-16:
What SKR model do you have?

Akeyr, 2023-06-17:
> What SKR model do you have? I did not notice - SKR mini E3 V2
> 
> Your board has one EXP3 connector. The TFT uses two connectors EXP1 and EXP2 Bad combination of HW.
> 
> Are you not using Marlin simulated mode, or are you using Y cable?

I use touch screen mode, not marlin mode.

radek8, 2023-06-17:
ok